stm32duino/Arduino_Core_STM8 #4: Incorrect cpp flags in platform.txt

adamberlinger, 2018-10-04:
Hello,
I tried to compile simple blink example, but I received a lot of errors. It seems the issue is in command line options being malformed. In `platform.txt` file there should be:
```
compiler.cpp.flags= {compiler.extra_flags} -I "{compiler.path}\hstm8++" -I "{compiler.path}\hstm8" {compiler.stm.extra_include} {compiler.warning_flags}
```
instead of:
```
compiler.cpp.flags= {compiler.extra_flags} -I "{compiler.path}\hstm8++" -c"i{compiler.path}\hstm8" {compiler.stm.extra_include} {compiler.warning_flags}
```
After changing the `platform.txt` file the compilation works perfectly.

VVESTM, 2018-10-04:
Hello, 
On my side, I don't have any problem with this syntax and I can compile examples without any issue.

On which version of Windows are you testing ? With which version of Arduino IDE ?
On my side, I am on Windows 7 with Arduino IDE 1.8.6 

hstm8++ directory contains C++ specific headers whereas hstm8 is dedicated to C. Can you give us the errors you have when compiling ? Is it a missing file ? With your modification, it can use a C header file for C++ code...

Thanks

adamberlinger, 2018-10-10:
Hello,
I'm using Windwos 7, and Arduino IDE 1.8.7.
The command-line command looks like:
```
"C:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\cxppstm8.exe" -gh -si -sd +proto +split -I "C:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8++" "-c\"iC:\\Users\\adam" "berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8\"" -I "C:\\Users\\adam berlinger\\Documents\\Arduino\\sketch_oct04a" -I "C:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Sxx_StdPeriph_Driver" -I "C:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Sxx_StdPeriph_Driver\\inc\\" -I "C:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Sxx_StdPeriph_Driver\\src\\" -I "C:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino\\stm8\\" -dSTM8Sxx -dF_CPU=16000000 -dARDUINO=10807 -dARDUINO_Nucleo_8S208RB -dARDUINO_ARCH_STM8 "-dBOARD_NAME=\"Nucleo_8S208RB\"" -d__arm__ -dSTM8S208 "-IC:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino" "-IC:\\Users\\adam berlinger\\AppData\\Local\\Arduino15\\packages\\STM8\\hardware\\stm8\\0.1.0\\variants\\NUCLEO_8S208RB" -sp -o nul "C:\\Users\\ADAMBE~1\\AppData\\Local\\Temp\\arduino_build_47853\\sketch\\sketch_oct04a.ino.cpp"
berlinger\AppData\Local\Arduino15\packages\STM8\tools\cxppstm8\4.1.3\\hstm8":
```
the output first prints some part of the command line arguments:
```
-I:

C:\Users\adam berlinger\Documents\Arduino\sketch_oct04a:

-I:

C:\Users\adam berlinger\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\system\Drivers\STM8Sxx_StdPeriph_Driver:

-I:

C:\Users\adam berlinger\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\system\Drivers\STM8Sxx_StdPeriph_Driver\inc\:

-I:

C:\Users\adam berlinger\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\system\Drivers\STM8Sxx_StdPeriph_Driver\src\:

-I:

C:\Users\adam berlinger\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\cores\arduino\stm8\:

-dSTM8Sxx:

-dF_CPU=16000000:

-dARDUINO=10807:

-dARDUINO_Nucleo_8S208RB:

-dARDUINO_ARCH_STM8:

-dBOARD_NAME="Nucleo_8S208RB":

-d__arm__:

-dSTM8S208:

-IC:\Users\adam berlinger\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\cores\arduino:

-IC:\Users\adam berlinger\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\NUCLEO_8S208RB:

-sp:

-o:

nul:

C:\Users\ADAMBE~1\AppData\Local\Temp\arduino_build_47853\sketch\sketch_oct04a.ino.cpp:
```
and then:
```
C:\Users\ADAMBE~1\AppData\Local\Temp\arduino_build_47853\sketch\sketch_oct04a.ino.cpp:1:10: fatal error: 'Arduino.h' file not found

#include <Arduino.h>

         ^~~~~~~~~~~

1 error generated.
```

Maybe it is related to the space in the include path.
Thank you.

closed 2018-11-06 by VVESTM

VVESTM, 2018-11-06:
Hi,
This is fixed in master branch. Issue was due to the space in path.
The syntax to use is -c 'i"{compiler.path}\hstm8"'

commit a9b1e8a7 (2018-11-06): Fix #4, issue when using space in path

C compiler must be called with -cixxx option. If xxx contains spaces, it
must be called with -ci"xxx". When using this double quote solution and
space between c and i, this is the case. Else the space in xxx is interpreted
as a separator.
commit c3c07054 (2018-11-06): Merge pull request #5 from VVESTM/#4

Fix #4, issue when using space in path