stm32duino/Arduino_Core_STM8 #7: Support for STM8S103/003

mcer12, 2019-06-25:
Hello, is there any possibility to add support for the small stm8s103f3p6 and stm8s003f3p6 ?

VVESTM, 2019-06-25:
Hello @mcer12,
There is no plan do it on our side for now. Do you need this for a specific project ? 
Normaly, stm8s103 and stm8s003 are supported by the SPL we used. So adding those 2 boards should consist to add 2 new variants. You can use what was done for stm8s208 (Arduino_Core_STM8/variants/NUCLEO_8S208RB/).
Any PR is welcome...

mcer12, 2019-07-28:
I see this as very useful because the stm8s103 and  stm8s003 are arguably the most common and cheapest chips. I use https://github.com/tenbaht/sduino but it's not compatible with other libraries. I tried to add a new variant but it didn't compile so I gave up as I don't really know what I'm doing... I would leave this thread open if anyone would like to contribute.

rishabhrpg, 2019-08-09:
do this library support stm8s003 ? as of now if not can it add?

VVESTM, 2019-08-19:
As mentioned previously, no plan to add new board on our side for now. Any help and PR is welcomed.

fpistm, 2020-02-24:
Added to to the request list: #15

closed 2020-02-24 by fpistm

MadZoids, 2020-10-06:
> Hello @mcer12,
> There is no plan do it on our side for now. Do you need this for a specific project ?
> Normaly, stm8s103 and stm8s003 are supported by the SPL we used. So adding those 2 boards should [...]

is it the same methods like creating new board for STM32?

fpistm, 2020-10-06:
Hi @MadZoids 
yes it is quiet the same

rtek1000, 2023-03-25:
sduino already supports this variant, can't use this information?

Apparently this repository is no longer being updated, more than 3 years without improvement:

https://github.com/tenbaht/sduino