stm32duino/Arduino_Core_STM8 #9: interfacing nrf24L01 with STM 8

vbshightime, 2019-07-26:
I am modifying rf24 library to interface it with stm8l152r8 but while doing so i encountered following error
#error clnk C:\Users\Agnext\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\NUCLEO_8L152R8\lkfscript.lkf:1 symbol __Z5delaym not defined (libraries\RF24-STM\RF24.cpp.o )

#error clnk C:\Users\Agnext\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\NUCLEO_8L152R8\lkfscript.lkf:1 symbol __Z10CSNportSeth not defined (libraries\RF24-STM\RF24.cpp.o )

#error clnk C:\Users\Agnext\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\NUCLEO_8L152R8\lkfscript.lkf:1 symbol __Z9CEportSeth not defined (libraries\RF24-STM\RF24.cpp.o )

#error clnk C:\Users\Agnext\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\NUCLEO_8L152R8\lkfscript.lkf:1 symbol __Z6millisv not defined (libraries\RF24-STM\RF24.cpp.o )

#error clnk C:\Users\Agnext\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\NUCLEO_8L152R8\lkfscript.lkf:1 symbol __Z9TxSPIbyteh not defined (libraries\RF24-STM\RF24.cpp.o )

Kindly help me resolve this

VVESTM, 2019-07-29:
Is it possible to have more details about the library you are trying to port ? (Exact name in library manager ?)
As mentioned on [this](https://github.com/stm32duino/Arduino_Core_STM8) page, be careful to endianness. You can have troubles with some libraries.
Then, for your porting, you have to identify the faulty line in C++ source code. Is it a missing function once the code is translated to C code ?

vbshightime, 2019-08-01:
yes i am using RF24 library by tmrh20. i tried modifying library to use it with stm 8 like  stm peripheral included with the core  instead of SPI library. Using only preferred functions to be used in my use case. but still i got the errors mentioned in the issue

closed 2019-10-12 by vbshightime