stm32duino/Arduino_Core_STM8 #13: I2C issue on STM8L151K6 custom board

YingJongTian, 2019-11-24:
I have made STM8L151K6 custom board compatible with Arduino board,
I have implemented LED blinking, Serial print and ADC function.
But I have some issue on I2C.
I have configured in variant.h, variant.cpp and PeripheralPins.c with STM8L151K6.

In PeripheralPins.c file, I have declared as following for I2C part.

#if !defined(NO_HWDAC)
const PinMap PinMap_DAC[] = {
    {PB_4,  DAC, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_Mode_In_FL_No_IT, 0, 1, 0)}, // DAC_OUT
    {NC,   NP,    0}
};
#endif

//*** I2C ***

#if !defined(NO_HWI2C)
const PinMap PinMap_I2C_SDA[] = {
    {PC_0,  I2C1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_Mode_In_FL_No_IT, AFIO_NONE)},
    {NC,    NP,    0}
};
#endif

I have connected I2C device with PC0 and PC1 pin, then run I2C scanner example, but I see only "unknown error" messages.
I'd like to get the help for this issue.

Thank you.

YingJongTian, 2019-11-24:
Sorry.
I have missed to upload SCL pin configuration.
#if !defined(NO_HWI2C)
const PinMap PinMap_I2C_SCL[] = {
    {PC_1,  I2C1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_Mode_In_FL_No_IT, AFIO_NONE)},
    {NC,    NP,    0}
};
#endif

fpistm, 2019-11-29:
Hi @YingJongTian 
Did you used pullup on the I2C lines ?

YingJongTian, 2019-11-29:
Thank you for your reply.
I have connected pullup, but it is same.
I have found the issue.
It was in variant.h and variant.cpp file.
I made location of I2C pins of digitalPin[] array to Nucleo_L152RB's digitalPin[] array.
So I have solved the issue.
Why the position of digitalPin[] array is related with function of peripheral?

Regards.

fpistm, 2019-11-29:
> Why the position of elements in digitalPin[] array is related with function of peripheral?

I don't understand well what you would mean.
In variant.h you should define the pin number values of all your pins.
Those value is simply the index to get the PinName in digitalPin[].
Then the PinName is used to find the correct config for a peripheral in the PinMap_<peripheral>

Ex: If you PC0 pin umber is 5 then the PinName is stored in digitalPin[5].
Then it looks for digitalPin[5] value (in this case should be PC_0).
Finally the AF is get from this config.

YingJongTian, 2019-11-29:
I thought so.
But in really, I changed the order of pins in array, so I resolved that problem.
So I have checked Nucleo_STM8S208 array too. it is same with Nucleo_STM8L too.

fpistm, 2019-11-29:
OK.
I close this issue as you solved it.

closed 2019-11-29 by fpistm