stm32duino/Arduino_Core_STM8 #14: Support Mac

sabas1080, 2020-02-11:
Hi

Please add support for Mac

I can help you for testing

fpistm, 2020-02-12:
Hi @sabas1080 
as mentioned in the README.md:

> Cosmic compiler, allowing to compile c++ source code on stm8 family (http://www.cosmic-software.com/)
> For now, the solution is running on Windows only.

While no compiler available for this host it would not be possible.

sabas1080, 2020-02-12:
@fpistm can this help?

https://github.com/tenbaht/sduino/tree/development/sduino/tools/macosx

fpistm, 2020-02-13:
Unfortunately not.

The main advantage of the cosmic compiler is that it provides C++ compiler. In that case you are able to use Arduino library without converting it to C.
sduino is not able to build C++, you have to convert it in C.

fpistm, 2020-02-18:
While no C++ compiler available for STM8 for Linux/Mac this will not be possible.
I close it as considered as answered.

closed 2020-02-18 by fpistm

rtek1000, 2023-03-25:
https://tenbaht.github.io/sduino/developer/compiling-the-tools/

rtek1000, 2023-03-25:
https://github.com/stm32duino/Arduino_Core_STM8/issues/42