stm32duino/Arduino_Core_STM8 #15: [New Variant] request list

fpistm, 2020-02-24:
Hereafter the list of all new variants requested

**Board support requested:**
 - [ ] STM8S103/003 (#7)
 - [ ] STM8L050 (#11)

Any contribution is welcome, so do not hesitate to submit a PR.

NicoBocki, 2020-04-30:
Hello i started to integrade the [STM8l050J3](https://github.com/NicoBocki/Arduino_Core_STM8/tree/new_variant_stm8l05j3/variants/SOT8_STM8L05J3), but i have never done it before. And i have already an issue with the genrated  PeripheralPins.c: <pleas can you give me a hint weher the problem could be.

"#error cpstm8 .\variants\SOT8_STM8L05J3\PeripheralPins.c:69(12+3) invalid pointer initializer"

Thank you in advance

fpistm, 2020-04-30:
Hi @NicoBocki 
hard to tell what is the problem...
This line seems correct:
https://github.com/NicoBocki/Arduino_Core_STM8/blob/55356d92ca1f9d7dc9525b3bd509cae894261842/variants/SOT8_STM8L05J3/PeripheralPins.c#L69

Honestly I have no clue at this time. error can be populated or something like that.

whoim2, 2020-11-20:
Requesting support for STM8L15xG4.

LaZsolt, 2022-12-15:
Hi Everyone!

I created some new variants. NUCLEO_8S207K8, STM8-DISCOVERY, STM8SVLDISCOVERY, Generic STM8L050J3M3, Generic STM8L151G4U6, Generic STM8S103F3P6 and Generic STM8S105K4T6.
https://github.com/LaZsolt/Arduino_Core_STM8

In the compile tests (Blink.ino), some were successful, others were not.
**Boards** NUCLEO_8S207K8 and Generic STM8L151G4U6 compile done without errors.

**Boards** DISCOVERY_STM8S and Generic STM8S105K4T6 compile error is:
`#error cpstm8 C:\Users\......\STMicroelectronics\hardware\stm8\1.0.0\cores\arduino\stm8\PortNames.c:72(19+5) GPIOH undefined` and `GPIOI` too.

In my opinion the reason is a bit complex:
The condition in file https://github.com/stm32duino/Arduino_Core_STM8/blob/784261b369b407612acca8ac2d39e0d86f5f4468/cores/arduino/stm8/PortNames.c#L70-L74 not copletely exact beause in file https://github.com/stm32duino/Arduino_Core_STM8/blob/784261b369b407612acca8ac2d39e0d86f5f4468/system/Drivers/STM8Sxx_StdPeriph_Driver/inc/stm8s.h#L2570 macro was defined regardless of the type of the MCU, but the definition of `GPIOH` and `GPIOI` macros https://github.com/stm32duino/Arduino_Core_STM8/blob/784261b369b407612acca8ac2d39e0d86f5f4468/system/Drivers/STM8Sxx_StdPeriph_Driver/inc/stm8s.h#L2650-L2654 depends of the MCU.
If I edit the file `stm8s.h` and remove the MCU dependent conditions before `GPIOH` and `GPIOI` macros definitions then the test compile became cuccessful.

**Boards** DISCOVERY_STM8SVL and Generic STM8S103F3P6 compile error is:
`#error cpstm8 C:\Users\......\STMicroelectronics\hardware\stm8\1.0.0\variants\DISCOVERY_STM8SVL\PeripheralPins.c:96(21+17) AFIO_UART1_ENABLE undefined`
I couldn't find any definition of the `AFIO_UART1_ENABLE` macro in any of the source files. 
(I have only one idea where it should be: `stm8_def.h`)

**Board** `STM8L050J3M3` made a similar compile error:
```
#error cpstm8 C:\Users\......\stm8\1.0.0\variants\STM8L050J3M3\PeripheralPins.c:109(18+15) AFIO_SPI1ENABLE undefined
#error cpstm8 C:\Users\......\stm8\1.0.0\variants\STM8L050J3M3\PeripheralPins.c:89(21+17) AFIO_USART1ENABLE undefined
#error cpstm8 C:\Users\......\stm8\1.0.0\variants\STM8L050J3M3\PeripheralPins.c:89(12+5) USART undefined
#error cpstm8 C:\Users\......\stm8\1.0.0\variants\STM8L050J3M3\PeripheralPins.c:62(12+3) I2C undefined
```
Maybe these macro definitions are missing from file `PinAF_STM8Lxx.h`?
(At my idea this macro names mismatch reason is some pads have multiple ports connected inside the package, then we need to use the internal port names, eg. (PA2) and `AFIO_USART1_PORTA_ENABLE` in the `PeripheralPins.c`. Maybe I'll try it later.)

Anyone have any ideas how to eliminate this errors?

fpistm, 2022-12-16:
Hi @LaZsolt 
I will try to check that in january when got some free time.

Deoptim, 2023-03-26:
@LaZsolt 
Hello.
I have only a STM8S103F3 board and was I test only it.
To compile blink sketch I was change only two files from your source https://github.com/LaZsolt/Arduino_Core_STM8:
```
--- a/cores/arduino/stm8/timer.c
+++ b/cores/arduino/stm8/timer.c
@@ -66,6 +66,7 @@
        .prescalerLimit = bits_16,
        .toggle_pin = {.pin = NC},
        .configured = 0},
+#ifdef TIM3
       {.timer_id = TIM3_E,
        .irqHandle = 0,
        .irqHandleOC = 0,
@@ -73,12 +74,16 @@
        .prescalerLimit = bits_16,
        .toggle_pin = {.pin = NC},
        .configured = 0},
+#endif
+#ifdef TIM4
       {.timer_id = TIM4_E,
        .irqHandle = 0,
        .timer_mode = TIMER_RESERVED,
        .prescalerLimit = bits_8,
        .toggle_pin = {.pin = NC},
-       .configured = 0}};
+       .configured = 0},
+#endif
+  };
 
   /**
   * @brief  Find the first timer not used
@@ -115,10 +120,14 @@
       return TIM1_E;
     if (peripheral == TIM2)
       return TIM2_E;
+#ifdef TIM3
     if (peripheral == TIM3)
       return TIM3_E;
+#endif
+#ifdef TIM4
     if (peripheral == TIM4)
       return TIM4_E;
+#endif
   }
 
   /**
@@ -157,6 +166,7 @@
       g_timer_config[TIM2_E].configured = 1;
       break;
 
+#ifdef TIM3
     case TIM3_E:
       TIM3_DeInit();
       if (IS_TIM3_PRESCALER_OK((TIM3_Prescaler_TypeDef)prescaler) == 0)
@@ -169,7 +179,9 @@
       TIM3_Cmd(ENABLE);
       g_timer_config[TIM3_E].configured = 1;
       break;
+#endif
 
+#ifdef TIM4      
     case TIM4_E:
       TIM4_DeInit();
       if (IS_TIM4_PRESCALER_OK(prescaler) == 0)
@@ -182,6 +194,7 @@
       TIM4_Cmd(ENABLE);
       g_timer_config[TIM4_E].configured = 1;
       break;
+#endif
 
     default:
       break;
@@ -209,6 +222,7 @@
       g_timer_config[TIM2_E].configured = 1;
       break;
 
+#ifdef TIM3
     case TIM3_E:
       TIM3_DeInit();
       if (IS_TIM3_PRESCALER((TIM3_Prescaler_TypeDef)prescaler) == 0)
@@ -221,7 +235,9 @@
       TIM3_Cmd(ENABLE);
       g_timer_config[TIM3_E].configured = 1;
       break;
+#endif
 
+#ifdef TIM4
     case TIM4_E:
       TIM4_DeInit();
       if (IS_TIM4_Prescaler(prescaler) == 0)
@@ -234,6 +250,7 @@
       TIM4_Cmd(ENABLE);
       g_timer_config[TIM4_E].configured = 1;
       break;
+#endif
 
     default:
       break;
@@ -262,15 +279,19 @@
       g_timer_config[TIM2_E].configured = 0;
       break;
 
+#ifdef TIM3
     case TIM3_E:
       TIM3_DeInit();
       g_timer_config[TIM3_E].configured = 0;
       break;
+#endif
 
+#ifdef TIM4
     case TIM4_E:
       TIM4_DeInit();
       g_timer_config[TIM4_E].configured = 0;
       break;
+#endif
 
     default:
       break;
@@ -353,6 +374,7 @@
       g_timer_config[TIM2_E].irqHandleOC = irqHandle;
       break;
 
+#ifdef TIM3
     case TIM3_E:
       TIM3_DeInit();
       CLK_PeripheralClockConfig(CLK_PERIPHERAL_TIMER3, ENABLE);
@@ -364,6 +386,7 @@
       g_timer_config[TIM3_E].configured = 1;
       g_timer_config[TIM3_E].irqHandleOC = irqHandle;
       break;
+#endif
 
     default:
       return;
@@ -397,6 +420,7 @@
     g_timer_config[TIM2_E].irqHandleOC = irqHandle;
     break;
 
+#ifdef TIM3
   case TIM3_E:
     TIM3_DeInit();
     CLK_PeripheralClockConfig(CLK_Peripheral_TIM3, ENABLE);
@@ -408,6 +432,7 @@
     g_timer_config[TIM3_E].configured = 1;
     g_timer_config[TIM3_E].irqHandleOC = irqHandle;
     break;
+#endif
 
   default:
     return;
@@ -438,9 +463,11 @@
       TIM2_DeInit();
       break;
 
+#ifdef TIM3
     case TIM3_E:
       TIM3_DeInit();
       break;
+#endif
 
     default:
       return;
@@ -478,9 +505,11 @@
         channel = TIM2_CHANNEL_1;
         break;
 
+#ifdef TIM3
       case TIM3_E:
         channel = TIM3_CHANNEL_1;
         break;
+#endif
 #elif defined(STM8Lxx)
     case TIM1_E:
       channel = TIM1_Channel_1;
@@ -490,10 +519,12 @@
       channel = TIM2_Channel_1;
       break;
 
+#ifdef TIM3
     case TIM3_E:
       channel = TIM3_Channel_1;
       break;
 #endif
+#endif
       default:
         return;
         break;
@@ -644,13 +675,17 @@
         data = TIM2_GetCounter();
         break;
 
+#ifdef TIM3
       case TIM3_E:
         data = TIM3_GetCounter();
         break;
+#endif
 
+#ifdef TIM4
       case TIM4_E:
         data = (uint16_t)TIM4_GetCounter();
         break;
+#endif
 
       default:
         break;
@@ -684,13 +719,17 @@
         TIM2_SetCounter(value);
         break;
 
+#ifdef TIM3
       case TIM3_E:
         TIM3_SetCounter(value);
         break;
+#endif
 
+#ifdef TIM4
       case TIM4_E:
         TIM4_SetCounter((uint8_t)value);
         break;
+#endif
 
       default:
         break;
@@ -746,6 +785,7 @@
         }
         break;
 
+#ifdef TIM3
       case TIM3_E:
         if (channel == TIM3_CHANNEL_1)
         {
@@ -756,6 +796,7 @@
           TIM3_SetCompare2(value);
         }
         break;
+#endif
 
       default:
         break;
@@ -794,6 +835,7 @@
       }
       break;
 
+#ifdef TIM3
     case TIM3_E:
       if (channel == TIM3_Channel_1)
       {
@@ -804,6 +846,7 @@
         TIM3_SetCompare2(value);
       }
       break;
+#endif
 
     default:
       break;
```
But you must compile without Serial (Disabled (No Serial)) option because no space for that.
Only basic Blink has size **4194byte** after compile - the code not optimized for small MCU.

LaZsolt, 2023-03-26:
@Deoptim 
Thank you your answare. I'll rewiev your changes soon.

fpistm, 2023-03-30:
Close as the repo will be archived.

closed 2023-03-30 by fpistm