stm32duino/Arduino_Core_STM8 #19: "No such file or directory : ctags_target_for_gcc_minus_e.cpp"

NicoBocki, 2020-04-28:
Hello:
I have a big problem when I'm trying to compile a sketch with my hardware definition. I'm running into an error and i didin't understand the cause of problem.

`
clang++.exe: error: no such file or directory: 'preproc\ctags_target_for_gcc_minus_e.cpp'

clang++.exe: error: no input files

C:\Users\N B\Documents\Arduino\STM8Test\STM8Test.ino:1:10: fatal error: 'Arduino.h' file not found

#include <Arduino.h>

         ^~~~~~~~~~~

1 error generated.

unknown file type B\AppData\Local\Arduino15\packages\STM8\tools\cxppstm8\4.1.3\\hstm8"

unknown file type -I

unknown file type C:\Users\N B\Documents\Arduino\STM8Test

unknown file type -I

unknown file type C:\Users\N B\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\system\Drivers\STM8Lxx_StdPeriph_Driver

unknown file type -I

unknown file type C:\Users\N B\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\system\Drivers\STM8Lxx_StdPeriph_Driver\inc\

unknown file type -I

unknown file type C:\Users\N B\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\system\Drivers\STM8Lxx_StdPeriph_Driver\src\

unknown file type -I

unknown file type C:\Users\N B\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\cores\arduino\stm8\

unknown file type -dSTM8Lxx

unknown file type -dF_CPU=16000000

unknown file type -dARDUINO=10805

unknown file type -dARDUINO_SOT8_STM8L05J3

unknown file type -dARDUINO_ARCH_STM8

unknown file type -dBOARD_NAME="SOT8_STM8L05J3"

unknown file type -d__arm__

unknown file type -dSTM8L05J3_Value_line

unknown file type -IC:\Users\N B\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\cores\arduino

unknown file type -IC:\Users\N B\AppData\Local\Arduino15\packages\STM8\hardware\stm8\0.1.0\variants\SOT8_STM8L05J3

unknown file type -sp

unknown file type -o

exit status 1
Fehler beim Kompilieren für das Board SOT8_STM8L05J3.
`

fpistm, 2020-04-28:
It seems you have an issu in your hardware definition. I guess in the boards.txt.

NicoBocki, 2020-04-28:
Hello thank you for your fast reply, could it be possible that is the white space in my user name? I found an post with the same error message

here: https://github.com/arduino/arduino-builder/issues/266

VVESTM, 2020-04-28:
An issue related to spaces in user name has already been solved in main branch : #5

fpistm, 2020-04-28:
Good catch @VVESTM .
@NicoBocki your user name is "N B" that ks why you get these errors.
You have to aplly the fix provided by @VVESTM

NicoBocki, 2020-04-29:
Thank you @VVESTM this sloved my issue

closed 2020-04-29 by NicoBocki