stm32duino/Arduino_Core_STM8 #21: Error at compile the "I2C/master_reader_example"

xpix, 2020-07-09:
hello,

try to compile the example and get this one, can u explain how to fix this?

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Frank\OneDrive\Dokumente\Arduino\libraries -fqbn=STM8:stm8:Nucleo_8L152R8:upload_method=MassStorage,xserial=generic -ide-version=10813 -build-path C:\Users\Frank\AppData\Local\Temp\arduino_build_258740 -warnings=none -build-cache C:\Users\Frank\AppData\Local\Temp\arduino_cache_188078 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM8Tools.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\STM8Tools\0.1.0 -prefs=runtime.tools.STM8Tools-0.1.0.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\STM8Tools\0.1.0 -prefs=runtime.tools.cxppstm8.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\cxppstm8\4.1.3 -prefs=runtime.tools.cxppstm8-4.1.3.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\cxppstm8\4.1.3 -verbose C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\hardware\stm8\0.1.0\libraries\Wire\examples\master_reader\master_reader.pde
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Frank\OneDrive\Dokumente\Arduino\libraries -fqbn=STM8:stm8:Nucleo_8L152R8:upload_method=MassStorage,xserial=generic -ide-version=10813 -build-path C:\Users\Frank\AppData\Local\Temp\arduino_build_258740 -warnings=none -build-cache C:\Users\Frank\AppData\Local\Temp\arduino_cache_188078 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM8Tools.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\STM8Tools\0.1.0 -prefs=runtime.tools.STM8Tools-0.1.0.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\STM8Tools\0.1.0 -prefs=runtime.tools.cxppstm8.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\cxppstm8\4.1.3 -prefs=runtime.tools.cxppstm8-4.1.3.path=C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\tools\cxppstm8\4.1.3 -verbose C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\hardware\stm8\0.1.0\libraries\Wire\examples\master_reader\master_reader.pde
Using board 'Nucleo_8L152R8' from platform in folder: C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\hardware\stm8\0.1.0
Using core 'arduino' from platform in folder: C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\hardware\stm8\0.1.0
Detecting libraries used...
"C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\cxppstm8.exe" -gh -si -sd +proto +split -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8++" "-c\"iC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8\"" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\libraries\\Wire\\examples\\master_reader" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver\\inc\\" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver\\src\\" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino\\stm8\\" -dSTM8Lxx -dF_CPU=16000000 -dARDUINO=10813 -dARDUINO_Nucleo_8L152R8 -dARDUINO_ARCH_STM8 "-dBOARD_NAME=\"Nucleo_8L152R8\"" -d__arm__ -dSTM8L15X_HD "-IC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino" "-IC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\variants\\NUCLEO_8L152R8" -sp -o nul "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_258740\\sketch\\master_reader.pde.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
C:\Users\Frank\AppData\Local\Temp\arduino_build_258740\sketch\master_reader.pde.cpp:
-DARDUINO_LIB_DISCOVERY_PHASE:
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\cxppstm8.exe" -gh -si -sd +proto +split -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8++" "-c\"iC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8\"" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\libraries\\Wire\\examples\\master_reader" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver\\inc\\" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver\\src\\" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino\\stm8\\" -dSTM8Lxx -dF_CPU=16000000 -dARDUINO=10813 -dARDUINO_Nucleo_8L152R8 -dARDUINO_ARCH_STM8 "-dBOARD_NAME=\"Nucleo_8L152R8\"" -d__arm__ -dSTM8L15X_HD "-IC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino" "-IC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\variants\\NUCLEO_8L152R8" "-IC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\libraries\\Wire\\src" -sp -o nul "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_258740\\sketch\\master_reader.pde.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
C:\Users\Frank\AppData\Local\Temp\arduino_build_258740\sketch\master_reader.pde.cpp:
-DARDUINO_LIB_DISCOVERY_PHASE:
Error while detecting libraries included by C:\Users\Frank\AppData\Local\Temp\arduino_build_258740\sketch\master_reader.pde.cpp
"C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\cxppstm8.exe" -gh -si -sd +proto +split -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8++" "-c\"iC:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\tools\\cxppstm8\\4.1.3\\\\hstm8\"" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\libraries\\Wire\\examples\\master_reader" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver\\inc\\" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\system\\Drivers\\STM8Lxx_StdPeriph_Driver\\src\\" -I "C:\\Users\\Frank\\OneDrive\\Dokumente\\ArduinoData\\packages\\STM8\\hardware\\stm8\\0.1.0\\cores\\arduino\\stm8\\" -dSTM8Lxx -dF_CPU=16000000 -dARDUINO=10813 -dARDUINO_Nucleo_8L152R8 -dARDUINO_ARCH_STM8 "-dBOARD_NAME=\"Nucleo_8L152R8\"" -d__arm__ -dSTM8L15X_HD "-IC:\\Users\\Frank\\OneDrive\\Dokumente\\ [... cut ...] \ctags_target_for_gcc_minus_e.cpp" "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_258740\\sketch\\master_reader.pde.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
**unknown file type -DARDUINO_LIB_DISCOVERY_PHASE_258740\sketch\master_reader.pde.cpp:**
-DARDUINO_LIB_DISCOVERY_PHASE:

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\STM8\hardware\stm8\0.1.0\libraries\Wire  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Nucleo_8L152R8.

fpistm, 2020-07-16:
Ho @xpix 

I've tested with the latest Arduino IDE 1.8.13 and I've reproduced your issue:
[image]

Using the previous version 1.8.12, it works:
[image]

It seems this is link to this change on Arduino side:
https://github.com/arduino/arduino-cli/pull/633

A new argument is added to the command line: `-DARDUINO_LIB_DISCOVERY_PHASE`
but for the specific cosmic compiler used for the STM8 core "-D" is not the right syntax, it should be "-d" but I don't know well the reason for this new arguments. I will ask to @facchinm about this.

xpix, 2020-07-17:
> arduino/arduino-cli#633

Hello,

Sounds good, i‘ll try to compile this with the 1.8.12 IDE.

Many Thanks 
Frank

fpistm, 2020-07-17:
Welcome. It is under investigation on arduino side.

jmakurami09, 2020-09-12:
Was it fixed on the arduino side?

fpistm, 2020-09-12:
> Was it fixed on the arduino side?

Yes but as far as I know not in official release yet.

jmakurami09, 2020-09-12:
I confirm "LAST UPDATE 17 August 2020 7:24:0 GMT" build still has not fixed this.

fpistm, 2020-09-14:
In fact, I think the core need a small update to be compatible.
That's why I'm waiting the new release which integrate the fix on Arduino Side then I will be able to update and test properly.

fpistm, 2021-07-14:
Hi @xpix 
With Arduino 1.8.15 this issue is fixed as  `-DARDUINO_LIB_DISCOVERY_PHASE` is no more present thanks  https://github.com/arduino/arduino-cli/pull/838
So I close this issue.

closed 2021-07-14 by fpistm