stm32duino/Arduino_Core_STM8 #26: unknown file type -DARDUINO_LIB_DISCOVERY_PHASEild_296524\sketch\CapacitiveSensorSketch.pde.cpp:

noeldiviney, 2020-10-02:
[Stm8-Build.zip](https://github.com/stm32duino/Arduino_Core_STM8/files/5315444/Stm8-Build.zip)
The Sduino sketches build ok

Just started with the STM8

Any help much appreciated.

Best regards 
Noel Diviney

fpistm, 2020-10-02:
Hi @noeldiviney 
This is a duplicate of #21

We are waiting a new Arduino release including the fix to update the core.
As a workaround use the 1.8.12 version.

closed 2020-10-02 by fpistm

noeldiviney, 2020-10-02:
Hi Frederic

Thank you for the prompt reply.
I will install 1.8.12 and report back

Many thanks.

Noel.

Noel Diviney Ayrenet Pty Ltd, Unit 3,80-88 Rodeo Drive, Dandenong South, Victoria 3175, Australia. Phone +61 413 120 892 email noel@ayrenet.com or divner@hotmail.com

noeldiviney, 2020-10-03:
Hi Frederic

Yes version 1.8.12 resolves the issue of the -DARDUINO_LIB_DISCOVERY_PHASE
I also installed the arduino 1.9.0 beta version and that also fixes the -D definition problem

However:
I ran foul of the Licensing circus.
I am trying to get a product out the door so I gave up trying to figure out how to download the licensed binary and reverted to using SDCC.

What would be nice is being able to sign a Licensing agreement with STMicro as a one time event and then donload tools as required.
I am trying to develop install scripts to set up all of our Company Computers so that we can use STMicro products in an efficient manner.

I am involved as a Systems Development Engineer at http://eicon.xyz/
PCBA | PCB Design | Schematic Design | Circuit Board Design | Moulding<http://eicon.xyz/>
Mostly concerned with PCBA see us for PCB design, PCB assembly, Schematic design, Circuit board design, PCB prototype, Schematic capture, Electronic parts and production runs from a few to thousands of professionally finished PCBs. We also have excellent moulding partners giving us strong advantages we can pass on to you in the form of an economical moulded parts source.
eicon.xyz

I look forward to working with you on future Arduino St projects

Many thanks for your assistance and best regards

Noel

Noel Diviney Ayrenet Pty Ltd, Unit 3,80-88 Rodeo Drive, Dandenong South, Victoria 3175, Australia. Phone +61 413 120 892 email noel@ayrenet.com or divner@hotmail.com