stm32duino/Arduino_Core_STM8 #27: Unable to install Cosmic License

noeldiviney, 2020-10-12:
I have obtained a Cosmic Stm8 Compiler License.lic file.
When I try to install the License I seem to be in an endliess loop wherby I end up back where I started.

Is there a Tutorial on how to achieve the installation.

Thanks Noel Diviney

fpistm, 2020-10-12:
Hi @noeldiviney 

the license is managed by Cosmic compiler. You should contact Cosmic support: support@cosmic.fr

noeldiviney, 2020-10-12:
Hi Frederic,

With all due respect I am trying to use a STMicro STM8 Processor which needs a compiler to build the code.
The licensing of the COSMIC Compiler is an absolute nightmare.
Is there somebody in ST who can describe the steps required to complete a build.
I have obtained and installed a License as required.
I get a message to press the "Finish" button to return to the Application (suggesting the license is acceptable) only to end up where I started ie (Submit a license request).
I am pleading for some help.

Noel Diviney.

Noel Diviney Ayrenet Pty Ltd, Unit 3,80-88 Rodeo Drive, Dandenong South, Victoria 3175, Australia. Phone +61 413 120 892 email noel@ayrenet.com or divner@hotmail.com

closed 2020-10-14 by noeldiviney

noeldiviney, 2020-10-15:
My apologies Frederic. Sorry.

Because of some confusion on my part I had requested the "PRODUCT=LXSTM8FSE_2020",
whereas,  it should have been "PRODUCT=CXCBE, CXSTM8"

With prompt help from COSMIC I was able to get this resolved.

I now have requested a license an had it installed on a separate Computer with complete success.
My method was as follows.
1. Install arduino-1.8.12    ( arduino-1.8.13 fails with the undefined definition " -DARDUINO_LIB_DISCOVERY_PHASE" problem.
2. Install the STM8 Core using https://github.com/stm32duino/BoardManagerFiles/raw/master/STM8/package_stm8_index.json
3.Launch Arduino Ide and configure a build for "STM8_S208RBT6_CapSensor" on the "Nucleo_8S208RBT6" board.
4. Select "Sketch->Verify/Compile.
5. During the Compile a pop up GUI "Cosmic STM8 C++ Compiler Registration for Arduino" appears.
6. Fill in the Details required and Select/Press one of the "On the Web, By Email, Edit Email or Write to File"
    Buttons. The first three insist on using Mi...sft Edge, so I selected "Write to File" and License.txt as the filename.
7. Copy the contents of License.txt and send it via email to 
    email address    =  stm8_Free@cosmic.fr
    Subject              = STM8FSE_2020_License_Request
    Contents           = as above contents of "License.txt" file
8. Leave the Arduino build App running and await a response from Cosmic.
9. When you receive the License.lic file from Cosmic it has to be copied to
    (your_path)/arduino-1.8.12/portable/packages/ STM8/tools/cxppstm8/4.1.3/License folder.
10. Select the "You have received and saved the license file" tick box in the GUI and Press the "Install" button.
11. Just follow the prompts from here and you are done.

Thank you all for the assistance.
The issue is now closed,
Best regards,
Noel Diviney.

fpistm, 2020-10-15:
No worry @noeldiviney  😉
Thanks for sharing your feedback.