stm32duino/Arduino_Core_STM8 #30: xcopy script is failing on the error below.

jmakurami09, 2021-09-22:
I'm running on Win 7, Arduino 1.8.16

looks like this file is not recognizing some option tools\STM8Tools\1.0.0/win/massStorageCopy.bat
Does = specify a file name
or directory name on the target
(F = file, D = directory)?

fpistm, 2021-09-22:
Hi @jmakurami09 
Could you share the full log ?
The script is available here:
https://github.com/stm32duino/Arduino_Tools_STM8/blob/master/win/massStorageCopy.bat

jmakurami09, 2021-09-22:
I updated my original issue.

I am using a ST Nucleo-8S208RB dev board

jmakurami09, 2021-09-22:
> Hi @jmakurami09
> Could you share the full log ?
> The script is available here:
> https://github.com/stm32duino/Arduino_Tools_STM8/blob/master/win/massStorageCopy.bat

I have the script but the xcopy is failing, it appears to have an extra parameter that is not needed.

jmakurami09, 2021-09-22:
Looks like this was related to the "wmic's module - logicaldisk" - so I hardcoded the Device ID in the script and voila.

fpistm, 2021-09-22:
OK.
Maybe you have a device which cause some trouble with an "=".
I could not reproduce this the WMIC command is used since a while. This is the same command used for the STM32 core:
https://github.com/stm32duino/Arduino_Tools/blob/4bfe2102721da20ab2fa98126ea195cc52f54159/win/massStorageCopy.bat#L43-L48
which is largely used.

If you find a generic solution feel free to submit a PR then I will validate it on win 10.

closed 2021-09-22 by fpistm