stm32duino/Arduino_Core_STM8 #32: Issue with variant STM8L052C6T6

0x0fe, 2022-04-02:
Hello,
I create d a variant STM8L052C6T, basedon the parameters of the L152 already present, and i was quite hopeful since in effect the STM8L052 uses the ST lib of STM8L152.  I imported the pinmp for my variant from C:\Users\dev\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM8Tools\1.0.0\scripts\genpinmap\Arduino\STM8L052C6Tx
And i condiently ran a test program to blink PA2 on which i have a LED. This test runs fine on STVD, but sadly when compiled on Arduino the STM8 does not run. I can compile and flash with no error, but it doesnt run correctly.
In an effort to understand i did two tests : 

-confirm wether the flasher part works, by flashing a known working binary (generated from STVD) from it.
The result is that yes, the flasher works, if i flash a working binary then the board runs as it should. So the problem is the binary generated by Arduino, not the flasher.

-confirm wether i can generate a working binary from the arduino toolchain:
I directly added some working HAL code in hw_config.c, in the thope that it can run and so i can debug from here. Sadly even when placed early this code does not run, so the problem is below this point.

I am now out of idea on what could be wrong. I did not found where the linker file is stored and selected, i would assuem the linker file needs to be changed even tho the STM8L052 uses the same HAL as STM32L152.
Ay advice is welcome, so close to work, that is frustrating

0x0fe, 2022-04-02:
well, it turned out the issue was indeed related to linking.
I modified the linker file and now it seems to work (at least my test works)

```
#
# *****************STM8L052C6T6******************
#
+seg .vector -b0x8000 -m0x8000 -n .vector -cs code -k	# interrupt and rest vectors
+seg .const -a .vector -n .const			            # constants follow vectors
+seg ._construct_ -a .const -n .construct -k		    # constructor table
+seg .text -a .construct -n .text			            # code follow constants
+seg .eeprom -b 0x1000 -m 0x100	-cs eeprom		        # internal eeprom
+seg .bsct -b 0x0 -m 0x100 -n .bsct -cs data		    # internal ram
+seg .ubsct -a .bsct -n .ubsct
+seg .bit -a .ubsct -n .bit -id
+seg .data -a .bit -m 0xcff -n .data
+seg .bss -a .data -n .bss
crtspi0.sm8
@*
libcppsl0.sm8
libfsl0.sm8
libisl0.sm8
libll.sm8
libm0.sm8
+def __memory=@.bss
+def __endzp=@.ubsct
+def __stack=0x7ff
+def __startmem=@.bss
+def __endmem=0x5fe
+def __ctorstart=start(.construct)
+def __ctorend=end(.construct)
```

fpistm, 2022-04-07:
Hi @0x0fe 
Fine you solve your issue.
To generate the linker file you can use the STM8CubeMx:
https://www.st.com/en/development-tools/stm8cubemx.html

As you find the solution, I close the issue. Thanks for sharing 👍

closed 2022-04-07 by fpistm