stm32duino/Arduino_Core_STM8 #33: I cant install library for STM8L

41linea45, 2022-08-20:
I cant install library for STM8L:

> Downloading packages
Failed to install platform: STMicroelectronics:stm8.
Error: 13 INTERNAL: Error downloading tool STMicroelectronics:cxppstm8@4.1.3: Get "http://cosmicsoftware.com/Arduino/4.1.3/Cxppstm8.zip": 

How can install library?

fpistm, 2022-08-22:
Hi @41linea45 
I've tested and it works.
Maybe an issue with your proxy? Firewall? antivirus?...

41linea45, 2022-08-22:
> Hi @41linea45 I've tested and it works. Maybe an issue with your proxy? Firewall? antivirus?...

unfortunately, same result. closed windows firewall i try download in arduino ide.
https://i.hizliresim.com/m5tp42m.PNG

fpistm, 2022-08-22:
Unfortunately, it really seems to be a network issue on our side.
Is it at home or at your work? Computer is personal or not ?

41linea45, 2022-08-22:
> Unfortunately, it really seems to be a network issue on our side. Is it at home or at your work? Computer is personal or not ?

this issue continuos approximately 10 days.
I use personel pc and no restriction

fpistm, 2022-08-23:
This package is not hosted by us but by partner Cosmic. 
Maybe you get geographical restriction and so you are not able to download it.
On my side I can't do nothing. Sorry.

You should try to contact cosmic as stated in the README.md:

> If you have a question about compiler, you can also [send a mail to Cosmic support](mailto:support@cosmic.fr) : send a mail to Cosmic support

closed 2022-08-23 by fpistm