stm32duino/Arduino_Core_STM8 #34: Acces to low power modes and functions & RTC IRQ handler

0x0fe, 2022-10-21:
Hello,
the STM8L serie is pretty much designed for ultra low power design, sadly, this core doesnt provide any example of available APIs to use neither RTC, low power mode and IRQs.
Besides, when i tried to call halt(); from the hal, i get 

```
C:\Users\xxxx\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm8\1.0.0\system\Drivers\STM8Lxx_StdPeriph_Driver\inc/stm8l15x.h:2925:18: note: expanded from macro 'halt'
 #define halt() {_asm("halt\n");} /*!<Halt */
```
so, could we have minimum example for ultra low power modes and RTC, and also how we should implement the IRQs of the halt, in particular the RTC IRQ.

fpistm, 2022-10-24:
Hi,

you should try to contact Cosmic support to know why _asm is not supported: support@cosmic.fr

0x0fe, 2022-10-24:
For that part, i found out, it is a linking issue in arduino core apparently, _asm wont work in cpp files (nor in ino file) even if used inside a function declared as extern "C", however, when used it in a .c file then i can call it without error.

As for teh second part of the question, can we have at least a minimal example for active halt with RTC or interrupt wake up? i have tried to use the HAL example and it never wakes, besides the LCD is shut down, meaning its not in active halt, but in halt.

[image]

[image]

closed 2022-11-21 by fpistm