stm32duino/Arduino_Core_STM8 #41: Internal pull-down on STM8L052

0x0fe, 2023-02-21:
I need to clarify wether pull-down is possible on STM8L052 because i got conflicting informations from the HAL and from the RM.
The RM seems to tell that we can use internal pull-up or pull-down, with 400K and 10K:

[image]

the HAL on the other hand, explicitely mention we can only use pull-up

https://github.com/stm32duino/Arduino_Core_STM8/blob/784261b369b407612acca8ac2d39e0d86f5f4468/system/Drivers/STM8Lxx_StdPeriph_Driver/src/stm8l15x_gpio.c#L15

so, what is correct? Can we enable pull-down of 10K or 400K on an arbitrary GPIO?
Thank you in advance.

fpistm, 2023-02-21:
Looking at the RM, this is not link to GPIO but to Routing interface (RI).
[image]

closed 2023-02-21 by fpistm

0x0fe, 2023-02-21:
i see, so it would only apply to inputs routed to TIM capture or ADC?

fpistm, 2023-02-21:
> i see, so it would only apply to inputs routed to TIM capture or ADC?

Don't know. You should ask such question on ST community.