stm32duino/Arduino_Core_STM8 #42: Add Linux support using SDCC compiler

rtek1000, 2023-03-25:
Hi,

I found a project ([w1209-firmware](https://github.com/mister-grumbler/w1209-firmware)) that uses [SDCC compiler](https://sdcc.sourceforge.net/) to compile, using the make command, perform the commands:

[How to install sdcc on Ubuntu](https://howtoinstall.co/en/sdcc) 
- sudo apt-get update
- sudo apt-get install sdcc

w1209-firmware [Makefile](https://github.com/mister-grumbler/w1209-firmware/blob/master/Makefile):
```
##
## Slightly corrected version of auto generated makefile by CodeLite IDE
##
## Build variables
ProjectName            :=thermostat
SourceDirectory        :=.
BuildDirectory         :=./Build
ObjectSuffix           :=.rel
IncludeSwitch          :=-I
LibrarySwitch          :=-l stm8
OutputSwitch           :=-o 
LibraryPathSwitch      :=-L
SourceSwitch           :=-c 
LinkerName             :=/usr/bin/sdcc
LinkOptions            := --out-fmt-ihx -mstm8
OutputFile             :=$(BuildDirectory)/$(ProjectName).ihx
ObjectSwitch           :=-o 
MakeDirCommand         :=mkdir -p
IncludePath            := $(IncludeSwitch). $(IncludeSwitch)./include 

##
## Common variables
## CC and CFLAGS can be overriden using an environment variables
##
CC       := /usr/bin/sdcc
CFLAGS   := $(LibrarySwitch) -mstm8

##
## User defined environment variables
##
Objects=$(BuildDirectory)/ts.c$(ObjectSuffix) $(BuildDirectory)/display.c$(ObjectSuffix) $(BuildDirectory)/timer.c$(ObjectSuffix) $(BuildDirectory)/buttons.c$(ObjectSuffix) $(BuildDirectory)/adc.c$(ObjectSuffix) $(BuildDirectory)/menu.c$(ObjectSuffix) $(BuildDirectory)/params.c$(ObjectSuffix) $(BuildDirectory)/relay.c$(ObjectSuffix) 

##
## Main Build Targets 
##
.PHONY: all clean MakeBuildDirectory
all: $(OutputFile)

$(OutputFile): $(BuildDirectory)/.d $(Objects) 
	@$(MakeDirCommand) $(@D)
	@echo "" > $(BuildDirectory)/.d
	$(LinkerName) $(OutputSwitch)$(OutputFile) $(Objects) $(LinkOptions)

MakeBuildDirectory:
	@test -d $(BuildDirectory) || $(MakeDirCommand) $(BuildDirectory)

$(BuildDirectory)/.d:
	@test -d $(BuildDirectory) || $(MakeDirCommand) $(BuildDirectory)

##
## Objects
##
$(BuildDirectory)/ts.c$(ObjectSuffix): ts.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/ts.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/ts.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/display.c$(ObjectSuffix): display.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/display.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/display.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/timer.c$(ObjectSuffix): timer.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/timer.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/timer.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/buttons.c$(ObjectSuffix): buttons.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/buttons.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/buttons.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/adc.c$(ObjectSuffix): adc.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/adc.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/adc.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/menu.c$(ObjectSuffix): menu.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/menu.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/menu.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/params.c$(ObjectSuffix): params.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/params.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/params.c$(ObjectSuffix) $(IncludePath)

$(BuildDirectory)/relay.c$(ObjectSuffix): relay.c
	$(CC) $(SourceSwitch) "$(SourceDirectory)/relay.c" $(CFLAGS) $(ObjectSwitch)$(BuildDirectory)/relay.c$(ObjectSuffix) $(IncludePath)

##
## Clean
##
clean:
	$(RM) -r $(BuildDirectory)/
```

To upload the *.ihx file to the STM8S:

stm8flash -c stlinkv2 -p stm8s003?3 -w thermostat.ihx 

[stm8flash INSTALL](https://github.com/vdudouyt/stm8flash/blob/master/INSTALL)
- sudo apt-get install pkg-config libusb-1.0-0-dev
- git clone https://github.com/vdudouyt/stm8flash.git
- cd stm8flash
- make
- sudo make install

https://github.com/mister-grumbler/w1209-firmware

fpistm, 2023-03-25:
Same issue it does not managed C++.

closed 2023-03-25 by fpistm

rtek1000, 2023-03-26:
I found this comment:

> C++ requires more code and data memory than standard C

Source: https://sourceforge.net/p/sdcc/discussion/1864/thread/91e76da8/

Maybe that's why sdcc supports C.

rtek1000, 2023-03-26:
> The MCUs that SDCC compiles for arent powerful enough to easilly run C++ programs like x86 processors. Its not *impossible* to make a C++ compiler, but its a bad choice of language for many processors, even x86 in some cases.

Source: https://sourceforge.net/p/sdcc/discussion/1864/thread/91e76da8/

rtek1000, 2023-03-26:
I left a post there on the SDCC forum, about the implementation of C++, I hope it will be possible to add this support:

https://sourceforge.net/p/sdcc/discussion/1864/thread/1f66d841a2/